googlecreativelab/open-nsynth-super #2: "You will need to obtain or build your PCB before proceeding further with this guide."

FuzzyWobble, 2018-03-13:
Hello :)
What exactly do you mean by "obtain"?

undothis, 2018-03-13:
Are there any builders out there?

Also if there are only 4 sound sources how can there be 100,000 sounds is that because of the variations of the 4 sound sources ?

HelveticaScenario, 2018-03-14:
i just used macrofab.com to order a partially assembled (smt) pcb. took some fiddling around tho

mcnasby, 2018-03-14:
Out of curiousity, what did the cost come to for the board through macrofab? Also, would you be willing to share any of the files you adjusted for use with macrofab?

ensonic, 2018-03-14:
The page make sas one can order the PCB with components, but I can't find any link :/

kasperkamperman, 2018-03-14:
@HelveticaScenario would you mind sharing the other necessary files.

sqfmi, 2018-03-14:
You can get the plain PCB here https://www.tindie.com/products/11702/

axmonti, 2018-03-14:
Ordered the PCB from the above link (cheap!); also created a "pure" Digi-Key BOM:

https://drive.google.com/file/d/1Hz521jbr1duW_I0DMHjfOVAaWWFHkrFS/view?usp=sharing

Note that the BOM doesn't include case hardware (screws, standoffs, etc.) and the only rotary encoder in-stock has 24 detents (instead of 12) - still 12PPR

aaronortiz, 2018-03-14:
Maybe we need a tablet app for this!

sqfmi, 2018-03-14:
The KLBR 4 Audio Connector wasn't available on Digi-Key, so I replaced the footprint with [another audio connector](https://www.digikey.com/product-detail/en/cui-inc/SJ1-3523N/CP1-3523N-ND/738689) in my [fork](https://github.com/sqfmi/open-nsynth-super/tree/master/pcb_hardware/projects/open-nsynth-super). I created a Digi-Key cart as well with all the components [here](http://www.digikey.com/short/jj2rt4), swapped some of the components that were low/out of stock.

HelveticaScenario, 2018-03-14:
the cost of the printing, the smt components, and the labor cost to place them came out to $123.08, without shipping. This doesnt cover any other costs like the through hole components, raspberry pi, screen, case, etc. The modifications i made were actually done inside their interface and i cant find a way to export or share it. ive emailed their support team and hopefully they will get back to me.

sqfmi, 2018-03-15:
@axmonti good call! I updated the footprint to accept switch/no switch connectors

RichardBronosky, 2018-03-15:
Why does this require a PCB? This is insane. RPi + potentiometers + rotary encoders + Android device connected via BLE. Done.

HelveticaScenario, 2018-03-15:
[NSynthSuperMacrofab.zip](https://github.com/googlecreativelab/open-nsynth-super/files/1817418/NSynthSuperMacrofab.zip)
These are the files, let me know if they dont work

ailgun, 2018-03-16:
@aaronortiz you probably can use the NSynth M4L device with MIRA and have it on tablet anyway.

undothis, 2018-03-16:
For some reason the telephone number to buy the PCB’s is not working. Any thoughts?

213-746-7272

Best,

Ory

Sent from the future

aaronortiz, 2018-03-16:
@ailgun Can you point me to MIRA? I don't know what it is, sadly.

ailgun, 2018-03-16:
@aaronortiz it's cycling74's iPad application to mirror the Max patches you have. 
https://cycling74.com/products/mira

Even magenta uses it in it's 'ai jam' example, you can see it here;
https://github.com/tensorflow/magenta-demos/tree/master/ai-jam-ableton
or here; https://www.youtube.com/watch?v=ZRnbbtqxBEc

My idea was to mirror the NSynth Max for Live patch to the iPad via MIRA so it practically would be a tablet version NSynth Super [without the colours etc]. I don't have time right now but might try to actually implement it and share it here in a few weeks.

sqfmi, 2018-03-17:
You can now get the semi-assembled PCB here https://www.tindie.com/products/11733/

All SMD parts soldered, just need to bring your own pi + discrete components

NullMember, 2018-03-17:
@sqfmi your tindie page says it's just bare PCB. Are you sure SMD parts are soldered?

cameronaaron, 2018-03-17:
I'm wondering this as well! He is the seller though so I trust him

sqfmi, 2018-03-17:
@NullMember @cameronaaron sorry about that, looks like the tindie admins haven't published the listing yet, should be up shortly!

NullMember, 2018-03-18:
@sqfmi interestingly when I search NSynth word on tindie I can see semi-assembled version in results. But when I click the product it's redirect me to your store page

MariusLub, 2018-03-19:
Same for me here, i just ordered the PCB from squaro's link. I wanted the semi assembled one but i just realized It said "plain". What should i be expecting ?

ailgun, 2018-03-19:
@NullMember it doesn't appear when searched as well.
I wonder when will it be online?

cameronaaron, 2018-03-19:
@sqfmi Any ideas when this will be up?

w4ilun, 2018-03-19:
Looks like the link is now up, sorry about the delay!

https://www.tindie.com/products/11733/

sqfmi, 2018-03-19:
Looks like the link is now up, sorry about the delay!

https://www.tindie.com/products/11733/

axmonti, 2018-03-20:
So, while we're waiting for boards/parts, has anyone installed the _[Open NSynth Super audio pipeline](https://github.com/googlecreativelab/open-nsynth-super/tree/master/audio)_ and started to create sounds to load on NSynth? It would be great if there was an initial corpus of sounds to download (either the initial wav files, or a ready-to-load set) and play around with. Complex convolutions appear to take a couple of days to process on decent hardware.

axmonti, 2018-03-29:
@sqfmi Received your board today - looks great - thanks (nice business card as well). Strange that the USPS tracking number provided still indicates that a label has been created but not yet in the system (not sure others have received their board yet). If I knew it was coming, I would have released my Digi-Key BOM, but at least I'll have the parts for a weekend build.

closed 2018-04-30 by nsynthsuper