googlecreativelab/open-nsynth-super #4: What's the best way to manufacture the PCB?

sinosoidal, 2018-03-14:
Hi. I would love to have some recommendations on websites where to produce the PCB from the files provided.

nevvkid, 2018-03-14:
@sinosoidal You can use https://aisler.net/ for example. As well they will provide all the electronic components you need to build the PCB. To source the display / raspberry pi you can probably go to adafruit.com, watterott.com or pimoroni.com

sinosoidal, 2018-03-14:
@nevvkid thanks for your feedback!

sqfmi, 2018-03-14:
@sinosoidal you can also buy the PCB from here https://www.tindie.com/products/11702/

sinosoidal, 2018-03-14:
I have orderedd an hour ago!

yamill, 2018-03-15:
@sqfmi when do you plan on selling a partially assembled and fully assembled pcb?

jonmoshier, 2018-03-16:
There's a facebook group called 'Pusherman', mostly it's DIY Eurorack, but there is a thread there about NSynth. ~$50 for a PCB with the SMD components pre-soldered, shipping in the US and the Cortex M0 has been programmed.

yamill, 2018-03-16:
@jonmoshier Sweet. Thanks for the info! I just tried joining the FB group, still waiting to get access to the group 👍

sqfmi, 2018-03-17:
@yamill we've been trying to sort out the logistics, you can now get the semi-assembled board :) https://www.tindie.com/products/11733/ (PCB + SMD components soldered, you'll need to bring your own pi, etc.)

sqfmi, 2018-03-19:
Looks like the link is now up, sorry about the delay!

https://www.tindie.com/products/11733/

yamill, 2018-03-20:
@sqfmi Amazing! Thank you for coming through! I will definitely place an order.

yamill, 2018-03-20:
@sqfmi just placed my order for the semi assembled. is it safe to say i'll need the following to complete the board?

1 | Adafruit 1.3" OLED display
1 | Raspberry Pi 3
1 | GorillaPi Heatsinks
1 | SanDisk 64GB MicroSD card
6 | Potentiometers
4 | Rotary encoders PEC11R-4315F-N0012
1 | USB socket
1 | MIDI socket
1 | 3.5mm audio jack
1 | tall rpi gpio header

sqfmi, 2018-03-20:
@yamill you'll need everything you've listed except for the USB socket, which comes pre-soldered (the tiny pins can be tricky with a soldering iron)

MariusLub, 2018-03-20:
@sqfmi i bought the plain and semi assembled pcb yesterday on tindie but in the end i only need the semi assembled one. Can you cancel my order for the plain one ? And can i get the money back ? 

Thanks !

panfriedcharlie, 2018-03-30:
@sqfmi Just a note, I received my unpopulated board the other day and was impressed with the quality. Thanks so much!

jjsomewhere, 2018-04-10:
Can anybody tell me if there are alternative rotary encoders that would work, or a third supplier for the one specified? In the UK the main supplier specified in the BOM has a 17 week lead time, and the backup supplier has a minimum order of 150. Any suggestions welcome!

panfriedcharlie, 2018-04-11:
@jjsomewhere I'm just _guessing_ here that many PEC11R parts will work keeping in mind the important aspects. Considering the data sheet, you're definitely going to want a PEC11R-4 part. Both detent options 2 and 3 have the same pulse count.

The next designator in the part no. is shaft length (15, 20, 25, 30mm). One could reasonably assume that the shaft length would make little difference to the part's operation and would more have to do with whether the knob fit. So if you're fine running without a knob or with a knob that floats slightly above the case, you should be good with any length.

The next designator is the shaft style. F (flatted) if you are going to use a knob. K (knurled) if you aren't or don't care.

The BOM has been updated to mention that either the switch (S) or no switch (N) versions are compatible.

The final designator is the resolution. Keep in mind that when you play with resolution and detents, you're looking at varying degrees of responsiveness.

jjsomewhere, 2018-04-11:
@panfriedcharlie Thankyou for your detailed reply, very helpful. I'll look in more detail later but I *think* that means the components I'm looking at, which differ only in the detant, should work fine. Their detant is double the resolution compared to the one specified, but so long as that's only a UI thing and doesn't feed back to the circuit it shouldn't make any difference. But I'll look around a bit more later and explore the other options, and maybe order an alternative and test them both before soldering anything. Thank you again.

KyleWerle, 2018-04-11:
I have the NSynth Super running pure data right now. Can't imagine why it wouldn't be able to run another magenta experiment setup.

jjsomewhere, 2018-04-12:
Humm interesting @kylewerle, I have an NSynth PCB in the post, will experiment!

closed 2018-04-30 by nsynthsuper

elimuro, 2018-05-09:
@ailgun check out this page on the digikey.ca website https://www.digikey.ca/products/en?FV=ffeca563 . there are other versions that should work fine that have longer or shorter knob length if you are ok with that

ailgun, 2018-05-09:
@elimuro thanks a lot. I have checked these out but I have already 3D printed the large dials. Would any of these fit right into it? I can do some minor edits by maybe carving it etc. but I can't say I'm a very handy person so I feel like I should buy the exact thing :)

jjsomewhere, 2018-05-09:
@elimuro I went with PEC11R-4215F-S0012
​, but I haven't built the NSynth yet so can't confirm that it works.
​

fakemau5, 2018-05-10:
@elimuro @jjsomewhere I built it with PEC11R-4215F-S0012 too and can confirm they works. The only downsides:
* you'll need to bend or cut the 2 extra legs (unused) to fit the board
* due to the doubled encoder resolution (24 steps instead of 12) the instrument changing is triggered every 2 steps on the knob.

jjsomewhere, 2018-05-10:
@elimuro @jjsomewhere Ah ha, thank you @fakemau5, useful info. That's what I suspected. I might think about getting a PEC11R-43 with a different knob length, as not building quite yet. Then again it isn't critical.

jjsomewhere, 2018-05-10:
@elimuro @jjsomewhere @fakemau5 I just had another look around, Mouser in the UK (and so maybe elsewhere) have 66 of the PEC11R-4315K-S0012 in stock. This appears to be identical to the PEC11R-4315F-S0012 except that it had a knurled knob rather than a smooth one. I just ordered some.

ailgun, 2018-05-10:
@jjsomewhere thanks for the info. Do you have any idea would these fit in to the 3D printed large dials?

jjsomewhere, 2018-05-10:
@ailgun Sorry, I can't say for certain. My guess is probably not exactly, but they're both 6mm shafts so it won't be far off. But I'm sure there are plenty of other knobs out there that will fit a 6mm knurled shaft, or models to print from on somewhere like Shapeways? I'm just going to wing it and see what works when I get to it.

ailgun, 2018-05-10:
@jjsomewhere Thanks a lot, I think I'll go with these and see if they'll fit or not - I can maybe carve inside or something if they don't exactly fit.

Also, would these work PEC11R-4215F-N0024? I found these on another seller. 

Thanks a lot really and sorry for the hassle, this is an area I'm totally alien to!
--

jjsomewhere, 2018-05-10:
No problem, my background's software, so the electronics stuff is new to me as well. Bourns have a pdf on their website which interprets the product codes for the PEC11R series (bottom of page one). http://www.bourns.com/docs/Product-Datasheets/PEC11R.pdf Don't rely on my interpretation but from what I understand that one would probably *not* work. It has 24 detents per revolution and 24 pulses, whereas the specified encoder as 12 of both.

ailgun, 2018-05-12:
@jjsomewhere Thanks a lot for the information, it seems I'll be going with PEC11R-4315K-S0012's and can buy again by September if these don't exactly work.