googlecreativelab/open-nsynth-super #5: Colour touch LCD version

rphi, 2018-03-14:
Are you likely to publish schematics for the model in the demo video with the colour touch LCD display? Although I appreciate the model here is designed to be simpler to build, having such a large display makes the thing look even more awesome, so if it weren't too expensive to do I'd love to give that model a try.

Cheers,
Rob

benjigreig, 2018-03-14:
Agreed. I'd love to be able to build that version too.

DaveRGP, 2018-03-14:
Absolutely agree

supchikov, 2018-03-14:
Want to build version with touch screen. Please, share it.

RichardBronosky, 2018-03-15:
Please do not bring new touch screens into the world!

We discard millions of Android and Apple phones every year. When planning any Maker project that needs an interface, find a way to allow people to repurpose these displays. (I don't mean disassembling the display. I mean via USB, BLE, even GPIO on known devices.) It makes me sad to see that you ever even designed that complicated PCB touch grid.

nsynthsuper, 2018-03-16:
Thanks for your interest Rob, the unit you refer to with the touch screen was designed as an experience prototype for developing the interface and we have no plans to release the source files.

closed 2018-03-16 by nsynthsuper