googlecreativelab/open-nsynth-super #14: Raspberry Pi mounting

FHIM, 2018-03-29:
I've been going through the documentation to see how the Raspberry Pi needs to be mounted, but I couldn't find any pictures. Everything else, like the OLED, the enclosure and all the mechanical parts is very well documented (hats off!), but it seems to be missing for the R Pi.
The bill of materials only mentions an extended female GPIO socket. I assume there needs to be a male pin header as well in the BOM as well.
Is there a recommendation for which one goes where? Socket on the R Pi and the header on the nsynth or vice versa?

KyleWerle, 2018-03-29:
The USB ports go towards the potentiometers. If you try to put it in any other way the rPi will be extruding pretty far off the main board.

I currently have an updated laser cut file for exposing the USB ports on the rPi.

FHIM, 2018-03-29:
Ah, thanks for your quick response!

closed 2018-04-30 by nsynthsuper