googlecreativelab/open-nsynth-super #16: Directory discrepency

panfriedcharlie, 2018-03-30:
Hi! Great project! The build was really fun, hope I didn't mess it up. 😜So I wrote 'onss_1.1_full.img' to a 64GB card and tried to follow the instructions in the firmware installation section of the make guide. It tells me:

> Next, navigate to the firmware directory and run the install command:
> 
> $ cd /home/pi/open-nsynth-super-master/firmware/src
> $ make install

Problem is I can't find the directory 'open-nsynth-super-master' anywhere on the card. I found 'open-nsynth-super' -- which leads down the same hierarchy of directories -- but when I 'make install' from there, I get an error: "could not intialize the debug port."

Any thoughts? Thanks in advance!!

KyleWerle, 2018-03-30:
In the lite image the directory was also /home/pi/open-nsynth-super/firmware/src

The -master would only be relevant if you were downloading the source files on a fresh image as that is what the guide seems to be based on.

A problem may be that you must run make before make install?

`$ cd /home/pi/open-nsynth-super/firmware/src`
`$ make`
`$ make install`

panfriedcharlie, 2018-03-31:
@KyleWerle thanks so much for your reply. I tried running "make" and I get:

> make: Nothing to be done for 'all'.

I tried the lite build this time and still got after I tried "make install":

> Error: Could not initialize the debug port
> TargetName: stm32f0x.cpu
> Type: cortex_m
> Endian: little
> TapName: stm32f0x.cpu
> State: unkown
> adapter speed: 1001 kHz
> Info: SWD DPIDR 00000000
> Error: Could not initialize the debug port
> Error: Target not examined, will not halt after reset!
> TARGET: stm32f0x.cpu - Not halted
> in procedure 'reset' called at file 'openocd.cfg", line 22
> in procedure 'ocd_bouncer'
> 
> Makefile:40: recipe for target 'install' failed
> make: *** [install] Error 1

Any thoughts? A few notes:

I see a lightning bolt in the upper right-hand corner of the screen. I'm using a pi 3 specific power adapter. Could that cause such an issue?

Also, should the OLED light up at all when plugged into HDMI before firmware install? It does not, does that mean something is screwy or does it need the firmware and hardware controllers to work?

Thanks again!

KyleWerle, 2018-03-31:
The STM firmware is what connects all the controls and oled screen to the rPi. To utilize all of those you will need to write the STM firmware.

It sounds like you are having a connection issue between the STM board and your rPi. Possibly double check the board? Is it the correct STM chip? Solder bridges? There are 2 STM chips that should be compatible.

panfriedcharlie, 2018-03-31:
Thanks @KyleWerle! I reflowed the ICs and Pi socket. Still no luck. Time to sit down and have a really close look at everything! Thanks again for the advice.

KyleWerle, 2018-03-31:
Good luck!

panfriedcharlie, 2018-03-31:
@KyleWerle Okay, yeah, my dumb fault. After another reflow and some fastidious cleaning I was able to write the firmware. Seems to be working okay now...

SarahHopeB, 2018-04-05:
I'm at the same step now and having difficulty with the make process as well.  I'm also using the lite image. I was able to successfully run make, when invoking make install I receive an error:

_BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 23
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info: BCM2835 GPIO JTAG/SWD bitbang driver
Info: SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)

**Error: An adapter speed is not selected in the init script. Insert a call to adapter_khz or jtag_rclk to proceed.

Makefile:40: recipe for target 'install' failed
make: *** [install] Error 1**_

So, after some Googling, I added the line into the openocd.cfg file...
adapter_khz 1001

From looking at the post above and seeing that 1001kHz was the selected adapter speed.  But still, no dice... It's now exiting with:

_BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 23
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info: BCM2835 GPIO JTAG/SWD bitbang driver
Info: SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info: clock speed 1001 kHz

**Error: BUG: current_target out of bounds
Makefile:40: recipe for target 'install' failed
make: *** [install] Error 255**_

Long story short, would someone on this thread who had a successful firmware flash care to show me what the openocd.cfg file looked like?  To my eyes it looks like it's simply not the right *.cfg that I'm using as the TargetName: stm32f0x.cpu isn't being found??

SarahHopeB, 2018-04-05:
I'm wondering if this is any way related to the fact that I had to use a mirror site for installing openocd yesterday.... The version I am using is Open On-Chip Debugger 0.10.0 -- Could someone confirm which version of openocd ended up working?

SarahHopeB, 2018-04-05:
Resolved this issue, reference the first problem with openocd repo being down: https://github.com/googlecreativelab/open-nsynth-super/issues/22

Once re-installing openocd from the secondary source within the makefile, I was able to execute 'make install' and received the same error code as @panfriedcharlie ... 

Then after inspecting the PCB I found that the STM chip pin 1 is not located to the upper-left when facing text on the IC. Instead, it is registered with the indentation. After rotating the chip on the board I was able to flash the firmware.

I'm all set!

panfriedcharlie, 2018-04-05:
@SarahHopeB I ~almost~ had that same issue. I second-guessed myself and zoomed waaay in on the build photos to check the dot orientations. Still too lazy to look up the pins and check the schematics. Glad you were able to solve the problem!

KyleWerle, 2018-04-09:
I ended up lining the dot / orientation marker on all the parts with the very small extra line on the silkscreen for the ICs.

SarahHopeB, 2018-04-10:
That's a great idea. I would also add a larger diode cathode band to the
opti-isolator midi circuit.  I had swapped the orientation initially due to
my squinty eyes...

KyleWerle, 2018-04-11:
We just need open source telescopic retinal enhancements and we'd be good.

USERxUNKNOWN, 2018-04-17:
Hey. So I used this image and it is asking for a login and password.  Any ideas?

SarahHopeB, 2018-04-17:
Asking for login/password at which step?

KyleWerle, 2018-04-17:
The default user is **pi** , and the password is **raspberry**.

closed 2018-04-30 by nsynthsuper