googlecreativelab/open-nsynth-super #24: Odd grid behavior

axmonti, 2018-04-08:
After building, installing, and successively testing the board with the Raspberry Pi 3 model B connected to a monitor, I shut down the system and tested the board in standalone mode. For some reason, the y-coordinate becomes “stuck” on the third row from the bottom, and by stuck, I mean that it’s quite difficult (but not impossible) to coax the grid indicator on the panel to move up or down to another y-coordinate by pressing on another row for a few seconds (the x-coordinate works fine).

On a whim, I plugged in the HDMI cable again, and the grid started working normally (and back to ‘sticky’ grid behavior when the cable is unplugged). I tried the 64GB image, but it had the same issue. It’s almost as if there’s an imaginary line on the 3rd-last row that make it difficult to cross when nothing is plugged into the HDMI port (unplugging or plugging in the mouse and keyboard had no effect).

Had anyone had the same odd grid behaviour? Could it have something to do with the model B and/or video driver? I double-checked the grid behaviour right to the IC pins with a scope and can’t see anything unusual between working and not working modes, which makes me think it’s software-related.

SarahHopeB, 2018-04-10:
Hmm.. That sounds like a strange correlation between HDMI cable and the grid...  Do you notice any differences in ground (floating, maybe?) when the HDMI cable is plugged in?  Are you sure you don't have any bridged pins on the IC or the pull-up/down resistors near the grid?

axmonti, 2018-04-10:
I originally had a solder bridge between pins 11 and 12 of U2, but that also shorted out power to the MCU from U6, so it was fairly easy to track down and rectify. Based on the correct touch behaviour when the HDMI cable is connected, I didn't think there was any lasting harm to the chip (or any others), and U2 controls the x-axis, which appears to work just fine.  That being said, I'll order two replacement touch chips on my next Digi-Key order (just in case). I also checked every trace passthrough to every resistor (especially for row 3), and they're all ok - nothing shorted.

I also left the board alone long enough for the Pi's 'screensaver' to kick-in (even though there wasn't a display attached). When I tried the touch surface, it worked normally until I chose another sound, at which point, an interrupt 'woke up' the display and the 'cursor' became stuck on the 3rd last row again.

I'm currently modifying ParticleScreen to display diagnostics on the oled, but a floating ground didn't occur to me, I'll first have a look from that PoV tonight - thanks for the suggestions!

axmonti, 2018-04-11:
Ok, good thinking on the ground issue - if I connect an external ground to any ground point on the board or the Pi, the touch panel works perfectly. As soon as the ground is disconnected, the cursor becomes stuck on row 9. The same goes for my hand - as long as I'm touching a ground point, the panel works normally. I'm not at all sure why row 9 is the only row affected; it's connected to pin 7 of U3, and the other side of the pullup is connected to the capacitive trace - all in keeping with every other row. I re-checked component connected to ground for continuity and 'leaks' with nothing found. Any additional thoughts/things to try would be appreciated.