googlecreativelab/open-nsynth-super #27: OLED screen won't turn on.

USERxUNKNOWN, 2018-04-18:
I have tried resoldering the jumpers and the pins but nothing is powering it.  Nsynth makes noise when plugged into my MPC. So I know that at least works.  Any ideas? Is there a firmware for the OLED that I am maybe missing?

KyleWerle, 2018-04-18:
I don't think the MIDI goes through the STM and goes directly from the optocoupler input to the rPi.
I'm pretty sure it is the same for the OLED, but with the way the nsynth app works it may not refresh correctly without the Firmware for the STM? I'm not entirely sure.

You may need to double check that you have flashed your STM with the nsynth firmware.

`$ cd /home/pi/open-nsynth-super/firmware/src`
`$ make install`

You should get a confirmation similar to this after successfully flash the STM
```
** Programming Started **
auto erase enabled
Info : device id = 0x10006444
Info : flash size = 32kbytes
wrote 12288 bytes from file ../bin/main.elf in 0.607889s (19.740 KiB/s)
** Programming Finished **
** Verify Started **
verified 11788 bytes in 0.050784s (226.680 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1001 kHz
shutdown command invoked
```

Good luck!

nsynthsuper, 2018-04-19:
The OLED doesn't require any firmware itself, if it's connected and working then it should power up. Regarding the firmware, if you can hear a difference when you turn the encoders and potentiometers then you will know that the firmware is installed and running correctly. 

If you have a spare Arduino lying around then it may be worth using it to test the screen module you've got using the Adafruit example library.

acbakhit, 2018-04-20:
I too am experiencing no display on my OLED, this is actually the second display I'm trying too, I'm thinking chances are low having two bad displays? Any other way of troubleshooting the display?

tekfunk, 2018-05-07:
+1

My display is not working as well.  When I hook up a monitor to the Pi, I can see the simulated display and all the knobs are working.  Is there not something that we can run on the pi that can look for the display?

nsynthsuper, 2018-05-14:
Per the previous comment, it would be worth checking that the OLED is functioning correctly using an Arduino or other separate device. Following that, you should check for continuity and shorts on your ONS board.

rockycamp, 2018-05-14:
A note of caution with this part of assembly - some separation of the boards is advisable when soldering pins to the respective pads as this will avoid solder bridges that can get sandwiched between boards. That said, I have yet to get the OLED working. I have checked power and traced data and clock to the STM. I have also confirmed functionality of the OLED with arduino. Using 16Gb image. Firmware successfully flashed. Controls, audio and midi are all working.

WellDrestGhost, 2018-05-29:
I am having similar issues as stated above.  
I am using the 16GB LITE  version of the firmware disk image.  I have my Raspberry Pi 3 connected via HDMI to a monitor, as well as a keyboard.  When the Rpi boots up, I can see what should display on the Nsynth's OLED in the upper-left hand corner.  When I press keys to login and start the firmware installation, I am able to hear sounds (as if i pressed a key on a midi keyboard)  coming out of the 3.5mm audio jack.   I am able to turn the encoders, and the 6 potentiometers and move my finger around the touch pad with the display on my monitor responding correctly.  After installing the firmware (via the instructions) I no longer get audio when pressing the keyboard.  My OLED screen does not show a display (even after reseting) and I am not getting any response from a  plugged in MIDI keyboard or Audio from the 3.5mm jack.  I tried to get audio of the Rpi's audio jack, and still no response.  I cannot see any physical solder bridges or shorts, however I am not ruling that out.  Any ideas?