googlecreativelab/open-nsynth-super #29: MIDI over USB

nwesthoff, 2018-04-24:
I would like to send MIDI over USB from a [Traktor Pro Remix Deck](https://support.native-instruments.com/hc/en-us/articles/212536009-How-to-use-TRAKTOR-s-Step-Sequencer-with-a-TRAKTOR-KONTROL-F1-Controller) (in step sequence or one-shot mode) to be able to use the NSynth in a live environment while DJ'ing.

Is it possible to send MIDI over USB? Or do I _have_ to use the 5-pin DIN port?
I couldn't find anything about it in the docs, and would try myself, but I'm currently waiting for the PCB to arrive.

KyleWerle, 2018-04-24:
The NSynth can only send / receive MIDI over USB as a USB Host. It is less easy to send MIDI between two host computers. There are many ways to do what you want, but very few of them are easy to setup and none as easy as connecting the NSynth Super to your computer with a USB cable.

The easiest solution in this case is to use a USB to MIDI DIN Adapter.

nanu-c, 2018-05-15:
This is easly done by adding the ofxMidi library to the app

nanu-c, 2018-05-15:
I can expand the openframeworksapp to do that and it's only some 20 lines of code

nanu-c, 2018-05-15:
* get a version of ofxMidi that works with yout version of openframeworks for 0.9.8 
`$ cd /home/pi/opt/of/addons && git clone https://github.com/danomatika/ofxMidi/`
* add `ofxMidi` to ~/opt/of/apps/open-nsynth/open-nsynth/addons.make
* in  /home/pi/opt/of/apps/open-nsynth/open-nsynth/src/ofApp.h add at the beginning after the includes `#include "ofxMidi.h"` and  change line 35 to
`class ofApp : public ofBaseApp, public ofxMidiListener{`
* add at line 73 
```
               // the midi stuff
                void newMidiMessage(ofxMidiMessage& eventArgs);
                ofxMidiIn midiIn;
                ofxMidiMessage midiMessage;
```
* In  /home/pi/opt/of/apps/open-nsynth/open-nsynth/src/ofApp.cpp change after line 77 add at end of void ofApp::setup() -> adapt this to your needs

```
	// print input ports to console 
	midiIn.listPorts(); // via instance -> comment this line when done
	//ofxMidiIn::listPorts(); // via static as well
	
	// open port by number (you may need to change this)
	midiIn.openPort(1);
	//midiIn.openPort("IAC Pure Data In");	// by name
	//midiIn.openVirtualPort("ofxMidiIn Input"); // open a virtual port
	
	// don't ignore sysex, timing, & active sense messages,
	// these are ignored by default
	midiIn.ignoreTypes(false, false, false);
	
	// add ofApp as a listener
	midiIn.addListener(this);
	
	// print received messages to the console
midiIn.setVerbose(true);
```
* add at the end of ofApp.cpp
```
void ofApp::newMidiMessage(ofxMidiMessage& msg) {
		midiMessage = msg;
	// ofLog(OF_LOG_NOTICE,"channel: " + ofToString(midiMessage.channel) + " control: "
	// + ofToString(midiMessage.control) +" value: " + ofToString(midiMessage.value)+" pitch: " + ofToString(midiMessage.pitch)
	// +" velocity: " + ofToString(midiMessage.velocity));
		int idx;
		idx = midiMessage.pitch;
		if (msg.velocity == 127){
			synthMutex.lock();
			synth.on(idx);
			synthMutex.unlock();
		}
		else{
			synthMutex.lock();
			synth.off(idx);
			synthMutex.unlock();
		}
}
```
* in ~/opt/of/apps/open-nsynth/open-nsynth run `$ make -j4`
* stop nysth service `sudo service  open-nsynth stop`
* run compiled app `$ bin/open-nsynth`

hagan66, 2018-08-29:
Has anyone done this and prepared a ready-to-go image file? I have no recent programming experience but I would very much like my Nsynth to speak USB. :-)

gitunit, 2019-01-07:
@nanu-c thx for the code. there is a minor typo though. it should be "listInPorts()" instead of "listPorts" at least in the latest version of ofxMidi.

nanu-c, 2019-01-08:
it's different, you have to put a usbmidi device in the raspberry pi. Like a akai or so.
If you wan't to use your pcb as usbmidi it's not possible like that because I  think the usb connector on the pcb is only for power.

if you want use the touch/knobs as midi input and send it to a other machine there a 2 common ways to do it.
1. connect the rpi to your windows network

2. use either https://github.com/hiroyuki/ofxArtnet for Artnet or

osc depending what is supported py the program you try to use in windows.
 https://github.com/openframeworks/openFrameworks/tree/patch-release/addons/ofxOsc https://www.youtube.com/watch?v=TczI-tSOIpY

gitunit, 2019-01-08:
ok. i get it. thx for clarification. btw. im from vienna, too. saw your homepage. nice projects ;)