googlecreativelab/open-nsynth-super #31: Jack 3,5mm doesn't fit into the PCB

minijouin, 2018-04-28:
My PCB (ordered on Tindle) doesn't have the right holes (not like on the picture) to put the 3,5mm audio jack ! 

Does anyone had the same problem ? 

Thanks !

minijouin, 2018-04-30:
Are you sure ? Cause it's seems to be exactly the same ...

minijouin, 2018-04-30:
This is the good one to use ! 
http://www.conrad.fr/ce/fr/product/705866/Jack-35-mm-Embase-femelle-horizontale-Cliff-FC68133-Nombre-total-de-poles-4-stereo-noir-1-pcs/?ref=detview1&rt=detview1&rb=1

jditte, 2018-04-30:
The one in Squaro Engineering's DigiKey shopping cart is made by CUI and has pins that are symmetric with respect to the center of the audio plug. These pins do not fit the published PCB. The one in the original BOM is made by Lumberg and has asymmetric pins.

jditte, 2018-04-30:
I decided to order the Lumberg part instead. DigiKey didn't have it, but other suppliers (Newark, for example) do...

nafem, 2018-05-03:
As mentioned above, the Cliff FC68133 fits my PCB not the Lumberg part.
[http://uk.farnell.com/cliff-electronic-components/fc68133/stereo-jack-3-5mm-5pos-pcb/dp/2518190](http://uk.farnell.com/cliff-electronic-components/fc68133/stereo-jack-3-5mm-5pos-pcb/dp/2518190)

jditte, 2018-05-03:
The semi-assembled PCB that is offered by Squaro Engineering on Tindie? Then there must be multiple versions. It certainly didn't fit mine...

tekfunk, 2018-05-07:
Yea this Audio jack is hard to find in the US, I found it here:

http://www.newark.com/lumberg/klbr-4/phone-audio-connector-socket-3/dp/97K2709?ost=97K2709&ddkey=http%3Aen-US%2FElement14_US%2Fsearch

@jditte I also got the semi-assembled PCB for Squaro and this fits.

It fits, BUT I still cannot get sound.  getting this error:

Playback open error: -16, Device or resource busy

kaosbeat, 2018-11-24:
yes there seems to be some discussion on the audio jack. I just received my PCB from Squaro (ordered from Tindie) and tyhe one I got from the supplied BOM does not fit at all.

I got this one (DOES NOT FIT)
https://be.farnell.com/pro-signal/mj-352w-0/connector-receptacle-3-5mm-phono/dp/1267400?CMP=i-bf9f-00001000

holes look different on my PCB then they do on the PCB assembly manual website though...

jditte, 2018-11-24:
@kaosbeat As mentioned before, this is the one that worked on my PCB from Squaro (ordered from Tindie):

https://be.farnell.com/lumberg/klbr-4/receptacle-3-5mm-phono-tht-r-a/dp/1217016

but I don't know whether they have redesigned the PCB in the meanwhile...