googlecreativelab/open-nsynth-super #33: Potentiometers instead of rotary encoders ?

minijouin, 2018-04-30:
Hi !

I'm wondering if it could be possible to put some potentiometers instead of the 4 rotary encoders in the corners (the good references is impossible to find on the internet and I had the one with 2 more arms -or could I cut them ?- )

Thanks !

fakemau5, 2018-05-05:
If you mean you got the "S0012" variant of the encoders, you're good to go. They have a "push button" feature, so they have the two more arms that are going to be short-circuited when you press the button, but given the "button" feature of the encoders is not used you can safely cut or bend them to fit on the board, and they will perform fine!

minijouin, 2018-05-05:
Ok thank you so much !

closed 2018-05-14 by nsynthsuper