googlecreativelab/open-nsynth-super #34: Midi input troubleshooting

mgiuli, 2018-05-02:
I'm desperate for some help.  I completed build a few days ago and have been unable to resolve the issue on my own.  I cannot get any sounds to be made with my synth.  I ran a sound test through a command and was able to hear noise out of the headphone jack.  That leads me to believe that no Midi signal is reaching the device.  I tried using the 64 and 16 GB preconfigured images.  With both, everything else appears to function just by plugging the imaged card into my system.  The screen functions and shows an expected response when each dial is turned.  On the Midi connection, I have tried plugging in a keyboard with 5 pin end on each side (no USB involved) and also tried playing some notes out of a couple of DAWs installed on my PC and connecting with a USB to Midi cable.  Are there any diagnostics that I can run to see if a signal is even being received?  Is there something additional I needed to do in first setup of the software?  Perhaps something else basic that I missed?  I don't have much knowledge in this area, so I won't be offended by any suggestions.  Thanks in advance!

Wolfberti, 2018-05-09:
I have the same problem. All the Inputs are working fine, but I don't get any sounds. I double checked the Soldering for Bridges and measured for wrong Voltages, but couldn't find an error. Are there preexisting scripts to test the MIDI input? Has anybody resources how to write a python script on your own?

Wolfberti, 2018-05-10:
Now it gets a little weird. I connected the pi with a keyboard, a hdmi Display and headphones. Now i can Play with the keys from q to p. At least I now know that the dac works.

mgiuli, 2018-05-12:
I've given up for the most part.  It may be time to conclude that my board is damaged and I need to find a new one.  I did run the dmesg command in console and reviewed the log.  This line is the only one that jumps out at me: 

 **2.188288] systemd[1]: [/etc/systemd/system/open-nsynth.service:3] Failed to add dependency on alsa.restore, ignoring: Invalid argument**  

Can someone please confirm if that's normal or not?

Wolfberti, 2018-05-12:
You could try to build the OS from Scratch. I'm currently downloading the 16gb version, if this will fail i will try Building from scratch.

nsynthsuper, 2018-05-14:
Hi mgiuli & wolfberti – the preconfigured image will have the baud rate set correctly; this does indeed sound more like an issue with the MIDI part of your hardware build. The easiest way to test the MIDI is to send notes to all channels which you should be able to do via your DAW.

mgiuli, 2018-05-19:
I'm glad that worked for you.  I knew it was a long shot for me, but I tried it anyways without success.  Thanks again!