googlecreativelab/open-nsynth-super #39: make it work with raspberry pi 3 b+ and latest openframework

nanu-c, 2018-05-14:
# Set up the rpi 3 b+
* get latest raspbian as described on https://www.raspberrypi.org/downloads/raspbian/

* Do the things in section 3 and 4 from https://github.com/googlecreativelab/open-nsynth-super/tree/master/linux#3-prepare-the-os
# get nsyth from git 
```
$ wget https://codeload.github.com/googlecreativelab/open-nsynth-super/zip/master
$ unzip master && rm master
```

# Set up the firmeware
* follow the guide from here https://github.com/googlecreativelab/open-nsynth-super/tree/master/firmware#installation-guide ->  get  `en.stm32cubef0.zip`  from http://www.st.com/en/embedded-software/stm32cubef0.html and put in /home/pi/open-nsynth-super-master/firmware/
```
$ cd /home/pi/open-nsynth-super-master/firmware/
$ unzip en.stm32cubef0.zip 
```
* get the depencies
`$ cd /home/pi/open-nsynth-super-master/firmware/utils && sudo ./install_dependencies.sh`
-> this takes a while 

* don't do this `$ mv STM32Cube_FW_F0_V1.9.0 cube` it gives then the error 
`make: *** No rule to make target '../STM32Cube_FW_F0_*/Drivers/CMSIS/Device/ST/STM32F0xx/Source/Templates/gcc/startup_stm32f030xc.s', needed by '../bin/main.elf'.  Stop.`

* flash chip
`$ make install`
```
sudo /home/pi/open-nsynth-super/firmware/openocd/bin/openocd --file openocd.cfg --command "program ../bin/main.elf verify reset exit"
Open On-Chip Debugger 0.10.0 (2018-05-14-21:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 23
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
cortex_m reset_config sysresetreq
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 1001 kHz
Info : SWD DPIDR 0x0bb11477
Info : stm32f0x.cpu: hardware has 4 breakpoints, 2 watchpoints
Error: stm32f0x.cpu -- clearing lockup after double fault
Polling target stm32f0x.cpu failed, trying to reexamine
Info : stm32f0x.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f0x.cpu       cortex_m   little stm32f0x.cpu       halted
adapter speed: 1001 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
adapter speed: 1001 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
adapter speed: 4061 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x10006444
Info : flash size = 32kbytes
wrote 12288 bytes from file ../bin/main.elf in 0.614188s (19.538 KiB/s)
** Programming Finished **
** Verify Started **
verified 11788 bytes in 0.050718s (226.975 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1001 kHz
shutdown command invoked
```
# copy audio files
* https://github.com/googlecreativelab/open-nsynth-super/tree/master/linux#4-copy-audio-files
# Set up latest openframework from git

* get latest openframeworks from git
`$ cd ~ && mkdir opt/ && cd opt/ && git pull --depth=1 https://github.com/openframeworks/openFrameworks.git`
* rename the openframeworks folder to be compatible with manual
`$ mv openframeworks of`
*  get openframeworks dependencies
`$ cd of/scripts/linux/debian && sudo ./install_dependencies.sh`
* get libs for openframeworks
`$ cd ~/opt/of/scripts/linux &&  download_libs.sh `
* copy the files from https://github.com/googlecreativelab/open-nsynth-super/tree/master/app/open-nsynth to `$ mkdir  /home/pi/opt/of/apps/open-nsynth && cp -r /home/pi/open-nsynth-super-master/app/open-nsynth /home/pi/opt/of/apps/open-nsynth/`
* replace in the code `Poco::FastMutex` with `std::mutex` -> this is in various files 
* build the app in `~/opt/of/apps/open-nsynth/open-nsynth` with
`$ make -j4` -> this takes a while
* get the settings.json in the right place `https://github.com/googlecreativelab/open-nsynth-super/tree/master/linux#4-copy-audio-files`
* test if it's running with 
`$ bin/open-nsynth`
# get nsynth dependencies
`$ cd ~/open-nsynth-super-master/linux`
* edit ./nsynth-setup.sh and edit the section replace every `gstreamer0.10` with `gstreamer1.0`
```
install_deps() {
    apt-get update
    apt-get install -y i2c-tools python-smbus gdb-arm-none-eabi gcc-arm-none-eabi \
        git autoconf libtool make pkg-config build-essential \
        libcairo-dev gstreamer1.0-dev gstreamer1.0-x \
        gstreamer1.0-plugins-base-apps gstreamer1.0-alsa \
        libudev-dev libsndfile-dev libopenal-dev libssl-dev \
        gstreamer1.0-plugins-good gstreamer1.0-plugins-bad \
        gstreamer-plugins-base0.10-dev freeglut3-dev libasound2-dev \
        libxmu-dev libxxf86vm-dev libgl1-mesa-dev libglu1-mesa-dev \
        libraw1394-dev libudev-dev libdrm-dev libglew-dev libopenal-dev \
        libsndfile-dev libfreeimage-dev libcairo2-dev libfreetype6-dev \
        libssl-dev libpulse-dev libusb-1.0-0-dev libopencv-dev \
        libegl1-mesa-dev libgles1-mesa-dev libgles2-mesa-dev libassimp-dev \
        librtaudio-dev libboost-filesystem-dev
}
```
* remove the follow section:
```
setup_openframeworks() {
    if ! [ -e /home/pi/opt/of ]
    then
        echo "Fetching openFrameworks"
        mkdir -p /home/pi/opt
        (
            cd /home/pi/opt
            curl http://openframeworks.cc/versions/v0.9.8/of_v0.9.8_linuxarmv6l_release.tar.gz | tar -xzf -
            mv of_v0.9.8_linuxarmv6l_release of
            #sudo of/scripts/linux/debian/install_dependencies.sh
        )
    fi
}
```
* run it `$ sudo sh ./nsynth-setup.sh`
* reboot `$ sudo reboot`

nanu-c, 2018-05-14:
and it works ;)

nsynthsuper, 2018-05-17:
Thanks so much nanu-c – will leave this issue open for others to reference.

ellasayers, 2018-12-15:
@nsynthsuper @nanu-c  is it possible to make an image for this, I've been hitting a lotttt of snags bouncing between these two instruction sets and cannot always find the correct packages referenced. Please and thanks!