googlecreativelab/open-nsynth-super #43: Midi real-time bytes cause the app's midi thread to shut down

damian0815, 2018-05-28:
The Midi thread currently misinterprets "system real-time" messages (which may be inserted into the midi byte stream at any point) as malformed data and exits. This makes nsynth unusable in a production environment where midi clock messages are used to keep devices in sync.

ref: http://www.lim.di.unimi.it/IEEE/MIDI/SOT3.HTM "Real Time messages can be sent at any time and may be inserted anywhere in a MIDI data stream, including between Status and Data bytes of any other MIDI messages."

nanu-c, 2018-05-28:
Let's have a look at this updated file:
https://github.com/nanu-c/open-nsynth-super/blob/master/app/open-nsynth/src/MidiThread.cpp

damian0815, 2018-05-28:
@nanu-c huh. I guess that's another way to achieve the same thing. I prefer mine TBH :)

jditte, 2018-06-05:
My solution in the thread "Lots of missed MIDI notes" fixes this problem, too (in addition to interpreting Note On with zero velocity correctly and being able to handle running status).

PR #44 (2018-07-30): Filter out system real-time messages
Fixes #43
commit cb8feb37 (2018-07-30): Fix #43 by filtering out system real-time messages
closed 2018-07-30 by nsynthsuper