googlecreativelab/open-nsynth-super #47: nsynth_generate fails with Abseil error.

9600, 2018-05-30:
Followed audio generation instructions and everything went fine until I got to executing nsynth_generate which, with or without arguments, fails with:

andrew@blast:/data/Liles/nsynth/open-nsynth-super/audio/workdir$ nsynth_generate
WARNING:tensorflow:From /usr/local/lib/python2.7/dist-packages/magenta/models/nsynth/wavenet/masked.py:116: __init__ (from tensorflow.python.ops.init_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.initializers.variance_scaling instead with distribution=uniform to get equivalent behavior.
Traceback (most recent call last):
  File "/usr/local/bin/nsynth_generate", line 7, in <module>
    from magenta.models.nsynth.wavenet.nsynth_generate import console_entry_point
  File "/usr/local/lib/python2.7/dist-packages/magenta/models/nsynth/wavenet/nsynth_generate.py", line 28, in <module>
    tf.app.flags.DEFINE_string("npy_only", False, "If True, use only .npy files.")
  File "/usr/local/lib/python2.7/dist-packages/tensorflow/python/platform/flags.py", line 58, in wrapper
    return original_function(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/absl/flags/_defines.py", line 241, in DEFINE_string
    DEFINE(parser, name, default, help, flag_values, serializer, **args)
  File "/usr/local/lib/python2.7/dist-packages/absl/flags/_defines.py", line 81, in DEFINE
    DEFINE_flag(_flag.Flag(parser, serializer, name, default, help, **args),
  File "/usr/local/lib/python2.7/dist-packages/absl/flags/_flag.py", line 107, in __init__
    self._set_default(default)
  File "/usr/local/lib/python2.7/dist-packages/absl/flags/_flag.py", line 196, in _set_default
    self.default = self._parse(value)
  File "/usr/local/lib/python2.7/dist-packages/absl/flags/_flag.py", line 169, in _parse
    'flag --%s=%s: %s' % (self.name, argument, e))
absl.flags._exceptions.IllegalFlagValueError: flag --npy_only=False: flag value must be a string, found "<type 'bool'>"

Any suggestions as to a remedy would be much appreciated!

9600, 2018-06-07:
@jesseengel any thoughts on this?

9600, 2018-06-13:
@nsynthsuper ?

jesseengel, 2018-06-15:
Sorry it took so long to respond. I just tested with head of magenta and nsynth_generate seems to work fine.

I think the problem is that directions for setting up the environment in https://github.com/googlecreativelab/open-nsynth-super/tree/master/audio are old now. It's actually pulling from a stale branch of my personal repo.

It should be sufficient to just follow the install istructions at https://github.com/tensorflow/magenta and do `pip install magenta-gpu` or `pip install magenta` and then `pip install tensorflow-gpu`.

@nsynthsuper, want to verify and update the docs?

9600, 2018-06-17:
Many thanks, this did the trick!