googlecreativelab/open-nsynth-super #48: Help...

unrealone1, 2018-06-01:
Hi experts,
OK built my unit,,,,

I have loaded the SD card with the image , booted and get presented with this image of lots of colours but it goes no further , any advice please

[image]

[image]

haigarmen, 2018-06-01:
Looks like your RaspberryPi doesn't have enough power. Try using a power adapter with 5 volts at 2.5 amps. https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md

unrealone1, 2018-06-01:
Hi Haig,
I currently am running the power /psu  through the nsync board only,  do I need a secondary power adapter for the raspberry as well
Thanks

YANN DAVIES
ABCOM IT Solutions
Unit 3 Ashwyn Business Centre
Marchants Way
Burgess Hill
West Sussex
RH15 8QY
+44 1444 871 200
www.abcom.co.uk
PROUDLY PROVIDING IT SERVICES FOR OVER 20 YEARS

unrealone1, 2018-06-02:
Hi ...I still have the same image even when suitable power supply on the raspbery pi and the nysnc board.

themangoest, 2018-06-02:
This is a b d rpi install

unrealone1, 2018-06-02:
So I just need to reinstall the image onto the Pi? I am downloading the 64Gb version on a 128GB card , hopefully that should sort it. By the way When I install the image should it completely wipe the Pi O/S..
Is there any tutorials out there?

unrealone1, 2018-06-03:
Hi all,
I am running a power supply to both the raspberry pi and the synth board, I have installed onto the sd card the larger 64Gb image.
I am still getting the rainbow image (with lightning strike)
Is the fact that Ive formatted completly the sd card the issue here ?
Do I need o/s on the card or a bootable partition?
Thanks

ErikOostveen, 2018-06-09:
Hi - I'd try loading the 16G image and see if that boots fine. I am having kernel panic errors with the 64G image, however, the 16G image runs just fine. I am using a RPI 3 Model B V1.2

damian0815, 2018-06-18:
@unrealone1 do you have a rpi 3 b or a 3 b+? The b+ didn’t work for me (same issue) but the b did.

ErikOostveen, 2018-06-20:
The exact wording on the PI's PCB is as follows "Raspberry Pi 3 Model B V1.2" ... so "B", not "B+". I have given up on the 64G image and stick to the 16G. It's now time to process my own sounds :-)

unrealone1, 2018-06-20:
Hi Erik, thanks for coming back to me. I am just about to buy the correct version PI and Ill update you once I have it...Thanks