googlecreativelab/open-nsynth-super #52: Encoder pins don't match

biggytruck, 2018-06-24:
Since I can't find PEC11R-4315F-N0012-ND, I use PEC11R-4315F-S0012-ND as encoders(also works according to the instruction). However, the encoders have 5 pins with 3 on one side and 2 on the other, while there are only 3 cutouts on the board. I looked at the PCB image and found that the demo board has 3 pins as well. How do I deal with this? Can I just cut off the extra 2 pins of the encoders?

tako3, 2018-06-29:
I'm having the same problem😱

NullMember, 2018-06-30:
You can use it. PEC11R-4315F-S0012-ND have built-in switch which PEC11R-4315F-N0012-ND don't have. Extra 2 pins are switch pins.

biggytruck, 2018-07-01:
@NullMember So can I trim the extra 2 pins off?

NullMember, 2018-07-01:
@biggytruck yes you can :)