googlecreativelab/open-nsynth-super #58: Compatible with Raspberry Pi 2?

zacYP2, 2018-08-15:
Hi guys, I already bought a Pi 2 because I found it cheap on craigslist, but I later realised this project calls for a RP 3. Is there anyway I can make it work or did I just throw money out the window? Cheers!

gitunit, 2018-12-20:
@zacYP2 have you tried it?

zacYP2, 2018-12-21:
I have successfully rebuilt the lite (16gb) OS image and got it working on the RP2. I have still to test midi and audio, but the interface is working. I'm having trouble with the screen because I am cheap and didn't want to order the adafruit one, so until I make that work or buy the real one we won't know. I have to leave for 1 month but I will probably continue once I come back

À+
Zac

gitunit, 2019-01-09:
@zacYP2 i finally tried it out. without recompiling i've just wrote the 64gb version 1.1 to the sd card and it ran so far ok on the RP2. except that it only receives 0 and 127 messages but nothing in between. but this is another issue.