googlecreativelab/open-nsynth-super #59: Chinese OLED display instead of adafruit?

zacYP2, 2018-08-17:
I noticed in the hardware folder that display communicates via i2c, does that mean we could actually use a generic i2c OLED instead of the 20$ adafruit one? As long as we solder the 4 pins in the right place it should work, right?

notthetup, 2018-08-22:
Yes. I have tried this and it works. The pinout is all over the place so you'll need some jumpers.

zacYP2, 2018-09-19:
I have recieved my OLED, it is this model: 

https://www.aliexpress.com/item/TZT-1-3-OLED-module-white-color-128X64-1-3-inch-OLED-LCD-LED-Display-Module/32777866923.html

As you said, pinout is different, I managed to plug DATA, CLOCK, VCC and GND in their respective places, but it still doesn't work. Maybe this is because I'm lacking the RESET pin?

I also have a Nokia 5110 I though I could use, which has a reset pin, but it didn't work, only random lines. Maybe because I need to connect the jumper on the back to switch to I2C instead of default SPI?

notthetup, 2018-09-25:
I believe there is a way to reset it using an SPI command. The driver actually uses that. So it should be OK. Check your connection again and maybe you can try to drive the display with an Arduino or something to make sure it works.