googlecreativelab/open-nsynth-super #62: Unable to connect to STM32

alexdgourlay, 2018-09-26:
Ok, I am really confused. I had everything working (OLED, audio, touch, encoders) and then everything stopped. Now I’m back to trying to connect to the micro controller and am stuck at this error. Help!

[image]

alexdgourlay, 2018-09-26:
Have checked functionality of all rpi GPIO pins and continuity of rpi header.
I have checked the voltage at supply pins of the STM32 and get a reading of 5V, is this correct?

cnuty, 2018-11-23:
Hi,
I have exactly the same problem and report with the STM32. I posted yesterday about it (post #67). Waiting for comments...
Cnuty

kaosbeat, 2018-11-26:
@alexdgourlay are you saying, you had a fully working box and it then stopped working?
on mine the display works, but there is no sound, and no working encoders (I can select instruments with the keyboard though)
problem indeed related (same error on screen) as #67 
I also measure 5V across C7, it should be 3.3V I read
did you get your board from tindie?

alexdgourlay, 2018-11-26:
Yep, had everything working. Then came back to the board and had a blank OLED screen. Tried re-running the make file which failed. 

Weirdly I managed to run the make file successfully once more, after adjusting the position of the RPi but then stopped working again.

Board was bought from tindie yes.