googlecreativelab/open-nsynth-super #66: no reaction to encoders, pots, touch panel and midi signal

RoaX2, 2018-10-20:
Hi all,

I finished a semiassambled nsynth, but it doesn't work.

RPI is booting correctly, the display shows the grid (also a monitor when connected), but there's no respond to encoders, pots and touch panel. There's also no respond to midi signals and there's no sound.
Does somebody has any idea?
Thank you,
RoaX

kaosbeat, 2018-11-24:
did you flash the firmware to the STM32?

RoaX2, 2018-11-24:
No. I thought that I do not have to do anything like this if I use a semi-assambled board with a precompiled image (in this case the light image) and it should be out of the box. Am I wrong with that?
Thank you for your answer.

kaosbeat, 2018-11-24:
I assume you still have to flash the board, I just got mine semi-assembled, but no luck so far. Flashing does not even work

follow these steps
[https://github.com/googlecreativelab/open-nsynth-super#4-install-the-firmware](url)

RoaX2, 2018-11-25:
OK, thanks so far. I'll have a try next days.

troglodisme, 2018-12-14:
Hi @kaosbeat @RoaX2 good to see somebody is having the same issues.

I have assembled my own boards and pretty sure they are 100% correct, but I am failing to install the firwmare! As soon as I power the board I can see the pi is quickly booting up for a few seconds (it doens't asked me to login like detailed in the guide here), then it starts showing the grid on the little display and also on my monitor, but nothing else happens. No sound, no potentiometers.. nothing.

I think I might have to go back and install the firmware? Do you know how I can load the terminal from that stage? 

Thanks!

kaosbeat, 2018-12-14:
Hey, measure the voltage on de voltage regulators it should be 3.3v if not you have faulty regulators. A batch was produced having this issue and it will have the symptoms you describe.

troglodisme, 2018-12-15:
Hey thanks @kaosbeat. I read your posts on that issue, but I doubt it would be the same problem because I didn't buy the PCBs from tindie. 

I have measured U4, U5, U6 and I can see 3.3 on the top left pin and 1.15 on the top right pin, using the bottom middle pin a GND reference. Could you confirm that is the correct reading?

I have also posted [here](https://github.com/googlecreativelab/open-nsynth-super/issues/68) to see if I could go back to the terminal and try to reinstall the firmware? Would you be able to help on that? Total newbie here!