googlecreativelab/open-nsynth-super #67: make install problem

cnuty, 2018-11-22:
Hello,
I read all the preview posts about a "make install" problem. I reflowed the Pi interface connector and all the STM32 connections. Same error with the lite version. The message is the following:
....
_BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 23
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info: BCM2835 GPIO JTAG/SWD bitbang driver
Info: SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info: clock speed 1001 kHz
in procedure 'init' called at file "openocd.cfg", line 20
in procedure 'ocd_bouncer'

Makefile:40: recipe for target 'install' failed
make: *** [install] Error 1

Please, do you have an idea how I can fix it?
Thanks a lot,
Cnuty

ajmonkey, 2018-11-24:
I'm having the same probem too! I've tried to rebuild from scratch--by downloading the nsyth-super-master here's my error-log:
pi@raspberrypi:~/open-nsynth-super-master/firmware/src $ sudo mount -o remount,rw /
pi@raspberrypi:~/open-nsynth-super-master/firmware/src $ make install
sudo openocd --file openocd.cfg --command "program ../bin/main.elf verify reset exit"
Open On-Chip Debugger 0.10.0 (2018-03-07-13:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 23
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
cortex_m reset_config sysresetreq
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 1001 kHz
in procedure 'init' called at file "openocd.cfg", line 20
in procedure 'ocd_bouncer'

Makefile:40: recipe for target 'install' failed
make: *** [install] Error 1

When I start up the raspi with nsyth board the power comes on--the oled displays the grid--if I move my mouse--the Grid on the monitor/Raspi software works---it moves the cursor position on the OLED--when I try moving/changing the cursor position on the trackpad--nothing happens--will check soldering---no idea why I can't build from Makefile--any trouble shooting steps or testing steps would be greatly appreciated-thanks!

kaosbeat, 2018-11-24:
same problem here,
tried building myself, and used the lite image as well...

is this a hardware bug?

kaosbeat, 2018-11-25:
I did some debugging.

My hardware seems to be fine, I verified the pin connections to the RPi and they are OK. If you need to verify check at least these connections (not testing the pots etc... just the programming interface)
[image]

The RPi itself seems to be ok as well

I started hacking the openocd.cfg in the `~/open-nsynth-super/firmware/src` folder, this changes behaviour, but no luck so far (just more bugs) I just want a simple test routine to verify the connection to the STM32, but I don't know how this works... anybody?

thanks

kaosbeat, 2018-11-26:
When cranking up the debugger using the `-d 3`flag passed on the openocd in the `Makefile` I get lot's of extra output. Still not making a lot of sense to me though.
I added the `transports` command to the `openocd.cfg` before the init, and there it works fine.
first thing that is off seems to be this:
`Debug: 311 26 bitbang.c:470 bitbang_swd_read_reg(): No valid acknowledge: ack=5``

full output below

```
:~/open-nsynth-super/firmware/src $ sudo make install
sudo /usr/bin/openocd -d 3 --file openocd.cfg --command "program ../bin/main.elf verify reset exit"
Open On-Chip Debugger 0.10.0 (2018-03-07-13:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 5 command.c:544 command_print(): debug_level: 3
Debug: 14 5 options.c:181 add_default_dirs(): bindir=/usr/bin
Debug: 15 5 options.c:182 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 16 5 options.c:183 add_default_dirs(): exepath=/usr/bin
Debug: 17 5 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 18 5 configuration.c:42 add_script_search_dir(): adding /root/.openocd
Debug: 19 5 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 20 5 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 21 5 configuration.c:82 find_file(): found openocd.cfg
Debug: 22 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface bcm2835gpio
Debug: 23 5 command.c:143 script_debug(): command - interface ocd_interface bcm2835gpio
Debug: 25 5 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_jtag_nums'...
Debug: 26 5 command.c:323 command_new(): BUG: command 'bcm2835gpio_tck_num' does not have the '.usage' field filled out
Debug: 27 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tck_num'...
Debug: 28 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_tms_num' does not have the '.usage' field filled out
Debug: 29 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tms_num'...
Debug: 30 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_tdo_num' does not have the '.usage' field filled out
Debug: 31 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tdo_num'...
Debug: 32 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_tdi_num' does not have the '.usage' field filled out
Debug: 33 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_tdi_num'...
Debug: 34 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_swd_nums'...
Debug: 35 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_swclk_num' does not have the '.usage' field filled out
Debug: 36 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_swclk_num'...
Debug: 37 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_swdio_num' does not have the '.usage' field filled out
Debug: 38 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_swdio_num'...
Debug: 39 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_srst_num' does not have the '.usage' field filled out
Debug: 40 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_srst_num'...
Debug: 41 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_trst_num' does not have the '.usage' field filled out
Debug: 42 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_trst_num'...
Debug: 43 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_speed_coeffs' does not have the '.usage' field filled out
Debug: 44 6 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_speed_coeffs'...
Debug: 45 6 command.c:323 command_new(): BUG: command 'bcm2835gpio_peripheral_base' does not have the '.usage' field filled out
Debug: 46 7 command.c:364 register_command_handler(): registering 'ocd_bcm2835gpio_peripheral_base'...
Debug: 47 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_peripheral_base 0x3F000000
Debug: 48 7 command.c:143 script_debug(): command - bcm2835gpio_peripheral_base ocd_bcm2835gpio_peripheral_base 0x3F000000
Debug: 50 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_speed_coeffs 146203 36
Debug: 51 7 command.c:143 script_debug(): command - bcm2835gpio_speed_coeffs ocd_bcm2835gpio_speed_coeffs 146203 36
Debug: 53 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_swd_nums 25 24
Debug: 54 7 command.c:143 script_debug(): command - bcm2835gpio_swd_nums ocd_bcm2835gpio_swd_nums 25 24
User : 56 7 command.c:544 command_print(): BCM2835 GPIO nums: swclk = 25, swdio = 24
Debug: 57 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_bcm2835gpio_srst_num 23
Debug: 58 7 command.c:143 script_debug(): command - bcm2835gpio_srst_num ocd_bcm2835gpio_srst_num 23
User : 60 7 command.c:544 command_print(): BCM2835 GPIO config: srst = 23
Debug: 61 7 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_reset_config srst_only srst_push_pull
Debug: 62 8 command.c:143 script_debug(): command - reset_config ocd_reset_config srst_only srst_push_pull
User : 64 8 command.c:544 command_print(): srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
Debug: 65 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select swd
[... cut ...]
Debug: 315 26 bitbang.c:360 bitbang_exchange(): bitbang_exchange
Debug: 316 26 bitbang.c:538 bitbang_swd_run_queue(): SWD queue return value: 05
Debug: 317 26 command.c:626 run_command(): Command failed with error code 5
User : 318 26 command.c:687 command_run_line(): in procedure 'init' called at file "openocd.cfg", line 24
in procedure 'ocd_bouncer'
Debug: 319 26 command.c:626 run_command(): Command failed with error code -4
User : 320 27 command.c:687 command_run_line(): 
Makefile:40: recipe for target 'install' failed
make: *** [install] Error 1
```

cnuty, 2018-11-26:
Thanks a lot Kaosbeat for all this precisions. Today first - if I have time -:) (working) - I will try my hardware connections, specially powers and I2C, which are used to download firmware from pi. But I think problem is probably software...

kaosbeat, 2018-11-26:
@cnuty where did you get your PCB?

cnuty, 2018-11-26:
Hi Kaosbeat,
I purchase my PCB semi-assembled at Tindie (https://www.tindie.com/products/squarofumi/nsynth-super-semi-assembled-googles-ml-synth/)

kaosbeat, 2018-11-26:
same as me, I tried contacting them with this issue, no reply so far...

ajmonkey, 2018-11-27:
Wow! I got my pcb from the same place too! 

I haven't had time to double check my hardware--but think it'll be same as either cnuty or kaosbeat--looking for another micro sd card to try software reinstall...

cnuty, 2018-11-27:
Hi ajmonkey,
I tried to reinstall many times with the "big" version of software and the "lite" version too with the same results and the same error...

kaosbeat, 2018-11-27:
I have similar measurements, big problem indeed ...

cnuty, 2018-11-28:
Hi all,
Today with an ohmmeter, I checked the routing of Pi GPIOs (powers, I2C...) to STM32, OLED, according to schematics: all is ok. I checked too the routing of the 3 regulators TPS73133, according to datasheet and  Tindie Nsynth schematics: routing is correct. Marking of regulators is ok: "T34", for 3.3V. I reflowed the pins of the 3 regulators.
Nevertheless, their output deliver 5.1V, not 3.3V! One regulator broken is possible but all the three regulators broken, with the same correct routing? I don't understand that.
Here is the link to the datasheet of TPS73133:  http://www.ti.com/lit/ds/symlink/tps731.pdf

cnuty, 2018-11-28:
Hi all,
Now I have sent this mail to Squaro Engineering, which sell via Tindie my semi-assembled NSynth. Waiting a reply:  "I bought a few weeks ago a semi-assembled NSynth PCB. I have a problem to download the firmware into the STM32. But there is also an hardware problem: the three regulators TPS73133 delivers 5.1V instead of 3.3V, according to the NSynth schematics. How is it possible?. Notice several users have the same problem (hard & soft): read this post: https://github.com/googlecreativelab/open-nsynth-super/issues/67. Thanks a lot for your answer."

simplysnake, 2018-12-01:
I'm also having the same issue. I also purchased my board semi assembled from tindie.

cnuty, 2018-12-04:
Hi all,
No answer from Tindie... What can we do now?

simplysnake, 2018-12-04:
I also wrote the seller on Tindie and received no reply. I haven't measured my voltage at U4,U5, U6 yet but I did check at C7 and it was bad (4.8). I'm guessing those voltage regulators could be bad... it might be worth trying to resurface mount new good ones idk.

sqfmi, 2018-12-05:
Hello @simplysnake @cnuty @kaosbeat @ajmonkey, and any other Tindie customers prior to 2018-12-3

Thank you for the feedback, we have done further testing and have confirmed the issue. As suspected, the output is not at the right voltage level because our supplier provided the wrong parts for U4,U5,U6. Instead of the **TPS73133** which should output 3.3V, the boards were supplied with the **TPS73150**, which outputs 5V.

We have received a new batch from the factory that fixes this issue, **please let us know through Tindie's messaging with your order number, and we will send you a completely new board with the correct voltage regulators**. You may keep your existing board if you choose to, and simply replace the voltage regulators manually for an extra nsynth.

We apologize for the trouble and the inconvenience caused. If there are any further questions, please do not hesitate and reach out to us directly or through Tindie.

ajmonkey, 2018-12-15:
Curious, has anyone received their replacement board yet?

cnuty, 2018-12-16:
Hi ajmonkey,
Not received too. But I live in France. I mail to Tindie 7 days ago et they answered  the new board shipped.

kaosbeat, 2018-12-16:
Me neither (Belgium)

simplysnake, 2018-12-21:
No I haven't received mine either. I'm in the US.

cnuty, 2019-01-01:
Hi all,
And happy new year.
I received the new board yesterday. Now - when i'll have time - , I have to unsolder pots, encoders, display, and first to test software Pi interface connector. Maybe this connector is only sufficient to test software.
Give news when boards received and software tested. Thanks.

himijendrix24, 2019-02-14:
**Is the stm32 broken after powering on with the wrong regulators?**

I changed the voltage regulators on my first (faulty board) and measure 3.3V but the touch pad and pots do not work. 

It should be possible to change the stm32 and upload the firmware, right?
The touch ics (at42qt2120) should be fine since the can operate on 1.8-5.5V

T-Zema, 2019-07-26:
Hey, sorry for backing to life this topic, but I have no idea what's going wrong after make install firmware :(

sudo openocd --file openocd.cfg --command "program ../bin/main.elf verify reset exit"
Open On-Chip Debugger 0.10.0 (2018-12-11-16:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
BCM2835 GPIO nums: swclk = 25, swdio = 24
BCM2835 GPIO config: srst = 23
srst_only separate srst_gates_jtag srst_push_pull connect_deassert_srst
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
cortex_m reset_config sysresetreq
srst_only separate srst_nogate srst_push_pull connect_deassert_srst
adapter_nsrst_delay: 100
adapter_nsrst_assert_width: 100
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : SWD only mode enabled (specify tck, tms, tdi and tdo gpios to add JTAG mode)
Info : clock speed 1001 kHz
Info : SWD DPIDR 0x00000001
Error: Could not initialize the debug port
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f0x.cpu       cortex_m   little stm32f0x.cpu       unknown
adapter speed: 1001 kHz
Error: Could not initialize the debug port
Error: Target not examined, will not halt after reset!
TARGET: stm32f0x.cpu - Not halted
in procedure 'reset' called at file "openocd.cfg", line 22
in procedure 'ocd_bouncer'

Makefile:40: recipe for target 'install' failed
make: *** [install] Error 1

tried to resolder one voltage regulator, because he war overheating ;v
Now the temperature is ok, but still i have make install error :<
Anyone can show me please where to look for?

kaosbeat, 2019-07-26:
Did you measure the voltage input on the chip? It should be no more than
3.3V

I think I described the pins to measure somewhere in this thread

T-Zema, 2019-07-26:
Thanks for a tip, will try tomorrow ;)

T-Zema, 2019-07-27:
> Did you measure the voltage input on the chip? It should be no more than 3.3V I think I described the pins to measure somewhere in this thread
> […](#)

Yeah. I think it is a power issue... I got only U5 voltage regulator working properly. The others (U4 U6) won't give 3.3V don't know why :/ I guess now it's time to solve a mystery. 

I used this puppies https://www.ebay.com/itm/5PCS-TPS73133DBVR-IC-REG-LDO-3-3V-15A-SOT23-5-TPS73133-73133-TPS73133D-73133D-T/192825175164?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

kaosbeat, 2019-07-28:
Looks ok.
I solved the problem by using the power from the 3.3v pin on the rpi
itself....

Works perfectly now

K

T-Zema, 2019-07-29:
Something weird happens when I power MCU near C7 - raspberry shuts down :/