googlecreativelab/open-nsynth-super #68: Can't get to terminal to install the firmware

troglodisme, 2018-12-14:
I have finished assembling my boards and was excited to start installing the firmware but as soon as I powered the board (with the raspberry pi connected to both screen and monitor), the booting started without asking me to login (as described in the instructions) and after a few seconds both the little screen and the monitor were displaying the nsynth grid correctly. The touchpad, potentiometers and midi don't seem to work though, and most importantly no sound is coming out the minijack. 

I assume the firmware is not properly installed? I would like to go back to the terminal and try running the mount commands, but.. how do I get to back to the terminal??

Thanks!

kaosbeat, 2018-12-22:
Hey I just read [here](https://github.com/googlecreativelab/open-nsynth-super/issues/32#issuecomment-393398718) how to do it.

> To replace the file and recompile the application you will need access to the Raspberry Pi. Hook it up to an HDMI monitor and a USB keyboard and boot it. Initially, you will see a copy of the Open NSynth Super screen. Hitting the ESC key will get you to a login prompt. Use the username "pi" (without quotes) and the password "raspberry" to log in. As the SD card is mounted as read only by default, you will need write access by remounting it: "sudo mount -u remount,rw /".

Hope this works!

K

troglodisme, 2019-01-10:
Thanks @kaosbeat - I eventually found out randomly while I was trying to get this done! Do you know if there is a way to add that to the guide/description? I feel other people might benefit from it.