googlecreativelab/open-nsynth-super #70: broken voltage regulators (U4-U5-U6) (bad Tindie batch) - FIXED

kaosbeat, 2018-12-23:
Hi all,

For those of you having a faulty board with the wrong voltage regulators (U4-U6) good news. I patched the board without using any voltage regulators at all, but this is considered experimental. It's a hack that worked for me, but there might be unknown side effects.
Basically I just draw the 3.3V off the RPi power rail. I soldered a wire to the connector on top (it's the "square" connector), feed it through a hole in the PCB and patch it into the output connectors of the different 3.3V circuits...

[image]

I read you can pull up to 800mAh form the 3.3V rail, given a sufficient power supply to the Rpi. This seems plenty for what it needs to do... YMMV

it works, and it is stable so far, but I'm ordering the correct voltage regulators. There must be a good reason somewhere to use these... would like to know the reason by the way.

good luck

kaosbeat, 2018-12-23:
fixed, so it can be closed. People in #67 will have had this problem as well some other issues can be related. Tindie has shipped out replacement boards (but I have not yet received mine) and desoldering all the pot's and the screen might be a lot more work then desoldering 3 voltage regulators... So that is why this can be a good thing to try.

closed 2018-12-23 by kaosbeat

cnuty, 2019-01-01:
Hi all,
And happy new year.
I received the new board yesterday. Now - when i'll have time - , I have to unsolder pots, encoders, display, and first to test software Pi interface connector. Maybe this connector is only sufficient to test software.
Give news when boards received and software tested. Thanks.