googlecreativelab/open-nsynth-super #79: all controller is ok , but no sound

PantalaLabs, 2019-08-07:
Hi , 

Just got my unit. 
All controls runs fine. But no sound at jack.
Is there any command line to test the sound ?

Thank you.

JasonQRY, 2021-01-16:
Hi，

Me, too. My unit has all functions, only can't output sound. I try to figure out the problem. Now I try to replace the PCM512 chip.

Did you have already solved the problem ?

Thx

PantalaLabs, 2021-01-16:
Hi Jason.

I my case , the problem was the optocoupler (i cant remember now, but i think it is the 6n139) .
Replaced and everything worked fine.

JasonQRY, 2021-01-16:
> Hi Jason.
> 
> In my case , the problem was the optocoupler (i cant remember now, but i think it is the 6n139) .
> Replaced and everything worked fine.

Oh, thx for the answer ! I will also check my optocoupler. Your answer is really helpful!