googlecreativelab/open-nsynth-super #83: Upgrading hardware to 1.2 - wire encoder switches

TTRCmedia, 2019-11-01:
Hello!
I just finished my build, using the original PCB layout. As I did not get the encoders without momentary switch, I use encoders with switch. but I did not cut off the unnecessary legs, I just bent them upwards so they are lying on top of the PCB.
If I read the current version of the schematics correctly, the switches for using the "new" feature to store patches are connected between ground and GPIO pins. So would it be possible to use some thin wire to connect one pin of each encoder to ground and the other one to the correct GPIO pin, or are there other changes to the schematics to make this feature work?
Regards,
Danyel

Jonny0815, 2019-11-10:
having the same question

nervousapps, 2019-11-10:
Just wire the encoder switch pins to the corresponding GPIO pins and upgrade the firmware. It works great !

TTRCmedia, 2019-11-10:
Great! I´ll give it a try :)