googlecreativelab/open-nsynth-super #87: NSynth Samples lib

Fluqz, 2021-02-03:
Hey, I built a NSynth 1-2 years ago and always had the problem that id like to upload more or different samples, but I have a 15 year old windows computer, which will probably take weeks to process the audio. I also dont have linux installed. The [NSynth Sound Maker](https://experiments.withgoogle.com/ai/sound-maker/view/) is the only project I found with different audio files, but I have no idea how to upload them to my NSynth. 

Id really appreciate if anyone would share his audio files and maybe could give me some hints on how to upload them! 

Thanks in advance :)

where-is-brett, 2022-07-02:
Hey mate, I am training two sets of samples at the moment, you can find them [here](https://github.com/where-is-brett/nsynth-pipeline/tree/main/audio/workdir/audio_input) and [here](https://github.com/where-is-brett/open-nsynth-super/tree/master/audio/workdir). I am happy to share the generated audio files if you're still interested. Just need a place to host the files (they are large ~40GB).

Fluqz, 2023-04-18:
Hey dude, thanks for your reply. Im a bit late, but did you manage to use your audio files? I sure am interested in the files, if we find a way to share 40gb :D

where-is-brett, 2023-06-14:
Unfortunately not. I only had access to a newer machine which didn't support python 2 or TF1.5, so I had to tweak the nsynth training pipeline to make it run on python 3. I did manage to make it run on python 3 with the later TF versions but the training performance was really poor - would have run for a month on a A100 machine! This is probably a matter of GPU drivers as the same tweaked pipeline trained 3x as fast on Colab GPU (with Pro subscription).